TheDeanLab/navigate #914: perZ stack acquisition with photometric

AdvancedImagingUTSW, 2024-06-05:
Need to confirm that perZ works with Hamamatsu. In @SJShep hand's, it does not work with photometric.

AdvancedImagingUTSW, 2024-09-29:
@SJShep - any updates?

SJShep, 2024-09-30:
Hi Dean,

We had to move the microscopes and ran into a couple of computer hardware
issues. I am setting up our ASLM this week to run in linux. The PVCAM and
NI drivers are available so we expect it should work. I will keep you
posted.

Best,
Steven

AdvancedImagingUTSW, 2024-09-30:
Hey Steven,

I recently realized that the DCAM drivers are also now available on Linux. This is something that we have wanted for a long time... Many of the NGFF's perform much better on Linux...  Let us know if you run into any challenges with getting navigate up and going on a Linux system. We have launched it before, but lacked the camera drivers at the time. Everything seemed to work. 

Also, congrats on your paper!

All the best,
Kevin

SJShep, 2024-10-01:
Hi Kevin,

Installing and launching with synthetic hardware was easy. I will
update our experience using the PVCAM this week.

Best,
Steven

AdvancedImagingUTSW, 2024-12-13:
Hey @SJShep,

Any update on this and can we merge your fix?

Thanks,
Kevin

SJShep, 2024-12-13:
Hi @AdvancedImagingUTSW,

I have the instrument up and running well in Ubuntu 22.04 now. Sorry, it's taken so long to share  our changes. I will open a pull request to merge now.

Best,
Steven