TheDeanLab/navigate #921: Error message when starting software

EvoleneP, 2024-06-17:
**Describe the bug**
I the following error message when launching navigate and it does not launch. Happened twice in a row today so far then the software finally opened.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Traceback**
Please copy and paste the traceback provided in your Anaconda Prompt here.

**Logs & Configuration Files**
Please drag and drop the logs and configuration files here. These are located in your `.navigate` folder, and can be found easily be opening Navigate, selecting File, then `Open Log Files` and `Open Configuration Files`.

(navigate) C:\Users\Nikon>navigate
vvvvvvvvvvvvvvvvvvvvvvvvvvv Child Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvv
 Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\device_startup_functions.py", line 84, in auto_redial
    val = func(*args, **kwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\devices\filter_wheel\filter_wheel_asi.py", line 69, in build_filter_wheel_connection
    raise Exception("ASI stage connection failed.")
Exception: ASI stage connection failed.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 501, in _child_loop
    obj = initializer(*initargs, **initkwargs)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\model.py", line 116, in __init__
    devices_dict = load_devices(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\device_startup_functions.py", line 1404, in load_devices
    devices["filter_wheel"][device["type"]] = load_filter_wheel_connection(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\device_startup_functions.py", line 857, in load_filter_wheel_connection
    tiger_controller = auto_redial(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\device_startup_functions.py", line 100, in auto_redial
    raise exception
Exception
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^ Child Process Traceback ^^^^^^^^^^^^^^^^^^^^^^^^^^^
 vvvvvvvvvvvvvvvvvvvvvvvvvvv Main Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvvv
Traceback (most recent call last):
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Nikon\miniconda3\envs\navigate\Scripts\navigate.exe\__main__.py", line 7, in <module>
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\main.py", line 113, in main
    Controller(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 163, in __init__
    self.model = ObjectInSubprocess(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 383, in __init__
    assert _get_response(self) == "Successfully initialized"
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
Exception

AdvancedImagingUTSW, 2024-06-21:
As we discussed in our meeting, this is still a weird issue since it isn't showing up on our microscopes, many of which are equipped with the same hardware.

The only thing that I can think of is that perhaps there is a better way to terminate the serial communication instance with the TIger Controller. If for some reason Python does not close the serial port, it could be left in a hanging state, which would make it inaccessible when you subsequently try to open it again.

Just keep us in the loop. In the meantime, I assume power cycling the TIger Controller will help.

EvoleneP, 2024-06-21:
Hi Kevin

Yes will keep you posted. I always turn off the Tiger controller at the end of the day or if those issues appear consistently.

Best

Evolene Premillieu, PhD
Light Sheet Microscopy specialist
BioFrontiers Institute
University of Colorado Boulder
https://advancedimaging.colorado.edu/

[External email - use caution]

closed 2024-07-17 by AdvancedImagingUTSW