TheDeanLab/navigate #957: Autofocus + Multiposition

AdvancedImagingUTSW, 2024-08-01:
Request from @SJShep.

What would it take for us to create a feature sequence where we perform an autofocus measurement for every position in the multi-position table? We would...

- move to the first position in the multi-position table (e.g., pos_0)
- move to z[0]
- measure the autofocus
- set the `F_Start` position
- move to z[-1], 
- measure the autofocus, 
- set the `F_End` position. 
- Image the full z-stack.
- move to the second position in the multi-position table (e.g., pos_0)
- ...

To make it even better, we could evaluate if the tissue is present or not before running the autofocus routine...

- move to the first position in the multi-position table (e.g., pos_0)
- move to z[0]
- evaluate if tissue is present. 
- if tissue is present
   - measure the autofocus
   - set the `F_Start` position
   - move to z[-1], 
   - evaluate if tissue is present.
   - if tissue is present
      - measure the autofocus, 
      - set the `F_End` position. 
   - if tissue is not present
      - pass ( no need to run autofocus or set focus position )
 -  Image the full z-stack
- move to the second position in the multi-position table (e.g., pos_0)
- ...

AdvancedImagingUTSW, 2024-08-05:
@dpshepherd @SJShep 

According to @annie-xd-wang, this feature sequence will get you most of the way there. It does not check for the presence of tissue, but does everything else.

```
[{"name": PrepareNextChannel,},({"name": MoveToNextPositionInMultiPositionTable,"args": (None,None,None,),},{"name": CalculateFocusRange,},{"name": ZStackAcquisition,"args": (True,True,"z-stack",),},{"name": WaitToContinue,},{"name": LoopByCount,"args": ("experiment.MicroscopeState.multiposition_count",),},),]
```

AdvancedImagingUTSW, 2024-08-05:
This one ties it all together.  The first popup window shows the overall logic, but does not include what happens within the `DetectTissueAndReturn` feature. 
[image]
If you click on `DetectTissueAndReturn`, it will show the logic within it. Here, it is set to look at 2 planes, which is the same as the first and last z-positions. You can add your own function to do the segmentation. If it is `None` it is just an Otsu, but this is done on a per-image basis without knowledge of brightness history. It is thus not the best thing possible. 

For `Percentage`, 1 = 100%. This is the threshold amount of the image that needs to be occupied by tissue in order to return True. If True, it then runs the `CalculateFocusRange`, acquires a `ZStackAcquisition`, and then `WaitToContinue` which enables large stage movements without camera timeouts.

[image]

SJShep, 2024-08-06:
Thank you @annie-xd-wang @AdvancedImagingUTSW, I will try implementing on our end and let you know how it goes.

AdvancedImagingUTSW, 2024-08-19:
@SJShep & @dpshepherd  - Any luck?

AdvancedImagingUTSW, 2024-09-11:
Pending update, used as an opportunity to place the example in the documentation [here](https://thedeanlab.github.io/navigate/user_guide/features/example_feature_lists.html). If anything is not exactly correct, we will have to update it accordingly.

closed 2024-09-11 by AdvancedImagingUTSW