TheDeanLab/navigate #974: Software freeze on ctASLMv2 while in use

JinlongL, 2024-08-22:
In Develop branch, software stopped responding suddenly while clicking Acquire. Shows error:

```
(navigate) C:\Users\Spectral>navigate
2024-08-22 12:02:21,676 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:22,176 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:22,677 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:23,177 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:23,677 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:24,178 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
2024-08-22 12:02:24,679 - model - WARNING - sutter: SutterFilterWheel - Could not establish Serial Port Connection
Exception in thread live signal:
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 1087, in run_live_acquisition
    self.get_stage_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 806, in get_stage_position
    return self.active_microscope.get_stage_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\microscope.py", line 782, in get_stage_position
    temp_pos = stage.report_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\stages\sutter.py", line 180, in report_position
    ) = self.stage.get_current_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 197, in get_current_position
    raise UserWarning(
UserWarning: Encountered response b'>\xff\xf2\x02Z\xff\xff'. You need to power cycle the stage.
Exception in thread live Data:
Traceback (most recent call last):
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 973, in _bootstrap_inner
    self.run()
  File "C:\Users\Spectral\.conda\envs\navigate\lib\threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 911, in run_data_process
    self.end_acquisition()  # Need this to turn off the lasers/close the shutters
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\model.py", line 822, in end_acquisition
    self.active_microscope.end_acquisition()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\microscope.py", line 449, in end_acquisition
    self.stop_stage()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\microscope.py", line 769, in stop_stage
    self.central_focus = self.get_stage_position().get("f_pos", self.central_focus)
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\microscope.py", line 782, in get_stage_position
    temp_pos = stage.report_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\stages\sutter.py", line 180, in report_position
    ) = self.stage.get_current_position()
  File "C:\Users\Spectral\Documents\GitHub\navigate\src\navigate\model\devices\APIs\sutter\MP285.py", line 197, in get_current_position
    raise UserWarning(
UserWarning: Encountered response b'\xe3\xff*\x00[\r'. You need to power cycle the stage.
```

[model_debug.log](https://github.com/user-attachments/files/16715234/model_debug.log)

AdvancedImagingUTSW, 2024-09-03:
I started logging every command we send to and receive from the MP285. Even when we just stop a continuous acquisition, interrupt stage is called...

```
2024-09-03 11:18:43,858 - model - DEBUG - MP285: Connection established.
2024-09-03 11:18:43,988 - model - INFO - hamamatsu: HamamatsuOrca Initialized
2024-09-03 11:18:43,988 - model - INFO - hamamatsu: HamamatsuOrcaFusion Initialized
2024-09-03 11:18:44,048 - model - DEBUG - equipment_solutions: RemoteFocusEquipmentSolutions - Opening Voice Coil on COM: COM7
2024-09-03 11:18:44,049 - model - DEBUG - equipment_solutions: RemoteFocusEquipmentSolutions - Sending command: b'd0\r'
2024-09-03 11:18:45,300 - model - DEBUG - equipment_solutions: RemoteFocusEquipmentSolutions - Received command: b'>!214\n'
2024-09-03 11:18:50,302 - model - DEBUG - equipment_solutions: RemoteFocusEquipmentSolutions - Servo Engaged
2024-09-03 11:18:50,393 - model - DEBUG - sutter: SutterFilterWheel - Placing device In Online Mode
2024-09-03 11:18:50,413 - model - DEBUG - sutter: SutterFilterWheel - Initialized.
2024-09-03 11:18:50,414 - model - DEBUG - sutter: SutterFilterWheel - Moving to Position 0
2024-09-03 11:18:50,485 - model - DEBUG - sutter: SutterFilterWheel -  Placed in Default Filter Position.
2024-09-03 11:18:50,524 - model - DEBUG - synthetic: SyntheticZoom Servo Initialized
2024-09-03 11:18:50,883 - model - DEBUG - MP285: safe_write(b'c\r')
2024-09-03 11:18:50,915 - model - DEBUG - MP285: position_information = b'DQ\xff\xffzf\x02\x00F\x8d\xfa\xff\r'
2024-09-03 11:18:50,915 - model - DEBUG - sutter: MP-285 - Position: {'y_pos': -14282.0, 'x_pos': 6292.24, 'f_pos': -1789.28}
2024-09-03 11:18:51,801 - model - DEBUG - base: Changed Zoom to 36X
2024-09-03 11:18:51,801 - model - DEBUG - base: Zoom position: None
2024-09-03 11:18:51,801 - model - DEBUG - base: Zoom position: None
2024-09-03 11:18:51,801 - model - DEBUG - model: Change zoom of CTASLMv2 to 36X
2024-09-03 11:18:51,803 - model - DEBUG - MP285: interrupt_move() called
2024-09-03 11:18:51,811 - model - DEBUG - MP285: read(b'\r')
2024-09-03 11:18:51,812 - model - DEBUG - MP285: safe_write(b'c\r')
2024-09-03 11:18:51,843 - model - DEBUG - MP285: position_information = b'DQ\xff\xffzf\x02\x00F\x8d\xfa\xff\r'
2024-09-03 11:18:51,843 - model - DEBUG - sutter: MP-285 - Position: {'y_pos': -14282.0, 'x_pos': 6292.24, 'f_pos': -1789.28}
2024-09-03 11:18:56,092 - model - DEBUG - MP285: interrupt_move() called
2024-09-03 11:18:56,100 - model - DEBUG - MP285: read(b'\r')
2024-09-03 11:18:56,100 - model - DEBUG - MP285: safe_write(b'c\r')
2024-09-03 11:18:56,132 - model - DEBUG - MP285: position_information = b'DQ\xff\xffzf\x02\x00F\x8d\xfa\xff\r'
2024-09-03 11:18:56,132 - model - DEBUG - sutter: MP-285 - Position: {'y_pos': -14282.0, 'x_pos': 6292.24, 'f_pos': -1789.28}
2024-09-03 11:18:56,292 - model - INFO - hamamatsu: HamamatsuOrca - subarray_hpos, 352.0
2024-09-03 11:18:56,292 - model - INFO - hamamatsu: HamamatsuOrca - subarray_hsize, 1600.0
2024-09-03 11:18:56,292 - model - INFO - hamamatsu: HamamatsuOrca - subarray_vpos, 352.0
2024-09-03 11:18:56,292 - model - INFO - hamamatsu: HamamatsuOrca - subarray_vsize, 1600.0
2024-09-03 11:18:56,718 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-09-03 11:18:56,727 - model - DEBUG - sutter: SutterFilterWheel - Moving to Position 0
2024-09-03 11:18:56,871 - model - INFO - ni: Waveform Template Name: Default
2024-09-03 11:18:56,878 - model - INFO - ni: Waveform Expand Num = 1
2024-09-03 11:18:56,879 - model - INFO - ni: Waveform Repeat Num = 1
2024-09-03 11:18:56,913 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-09-03 11:18:56,914 - model - DEBUG - MP285: safe_write(b'c\r')
2024-09-03 11:18:56,932 - model - DEBUG - MP285: position_information = b'DQ\xff\xffzf\x02\x00F\x8d\xfa\xff\r'
2024-09-03 11:18:56,932 - model - DEBUG - sutter: MP-285 - Position: {'y_pos': -14282.0, 'x_pos': 6292.24, 'f_pos': -1789.28}
2024-09-03 11:18:57,163 - model - INFO - model: Navigate Model - Running data process, get frames [0]
2024-09-03 11:18:57,163 - model - INFO - model: Navigate Model - Sent through pipe0
2024-09-03 11:18:57,181 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-09-03 11:18:57,181 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-09-03 11:18:57,408 - model - INFO - model: Navigate Model - Running data process, get frames [1]
2024-09-03 11:18:57,409 - model - INFO - model: Navigate Model - Sent through pipe1
2024-09-03 11:18:57,426 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-09-03 11:18:57,426 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-09-03 11:18:57,655 - model - INFO - model: Navigate Model - Running data process, get frames [2]
2024-09-03 11:18:57,655 - model - INFO - model: Navigate Model - Sent through pipe2
2024-09-03 11:18:57,673 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-09-03 11:18:57,673 - model - DEBUG - feature_container: running signal node: LoopByCount
2024-09-03 11:18:57,901 - model - INFO - model: Navigate Model - Running data process, get frames [3]
2024-09-03 11:18:57,901 - model - INFO - model: Navigate Model - Sent through pipe3
2024-09-03 11:18:57,919 - model - DEBUG - feature_container: running signal node: PrepareNextChannel
2024-09-03 11:18:57,919 - model - DEBUG - feature_container: running signal node: LoopByCount
[... cut ...]
2024-09-03 11:19:12,062 - model - INFO - model: Navigate Model - Sent through pipe6
2024-09-03 11:19:12,062 - model - INFO - model: Navigate Model - Data thread stopped.
2024-09-03 11:19:12,062 - model - INFO - model: Navigate Model - Received frames in total: 7
2024-09-03 11:19:12,081 - model - DEBUG - MP285: interrupt_move() called
2024-09-03 11:19:12,087 - model - DEBUG - MP285: read(b'\r')
2024-09-03 11:19:12,087 - model - DEBUG - MP285: safe_write(b'c\r')
2024-09-03 11:19:12,087 - model - DEBUG - MP285: safe_write(b'c\r')

```

Here, it froze immediately...

AdvancedImagingUTSW, 2024-09-03:
@annie-xd-wang - See here: https://github.com/TheDeanLab/navigate/tree/stop_stopping_stage

zacsimile, 2024-09-04:
According to Jinlong's traceback, somehow the required number of bytes for the read (13 bytes, see line 188 of MP285.py) is getting split int 7 bytes in the live thread and 6 bytes in the data thread. My initial guess is the timeout isn't long enough for whatever reason to finish the 13-byte read, so it quits and then tries again in the other thread. Try setting the MP285 timeout to 1 to see if it fixes the issue.

AdvancedImagingUTSW, 2024-10-04:
Fixed with #1006