TheDeanLab/navigate #999: Simplify System Control - Dissemination Variant

AdvancedImagingUTSW, 2024-09-25:
*Overview*
As part of the UTD Capstone project, our goal is to simplify the optoelectronics of the microscope. Currently, we often have stage controllers, filter wheel controllers, an NI DAQ card, piezo controllers, etc. Each one of these things has to be independently wired up. This serves as an impediment for non-experts, slows adoption, and increases the likelihood that mistakes will require significant debugging.

*Requirements*
The microscope requires synchronous delivery of a number of analog and digital waveforms, serial commands, DLL calls, etc. Regarding the analog/digital control, we require:

Waveform type: triangle/sawtooth/sine/DC. Should enable control of amplitude, offset, phase.
Waveform duration: Camera exposure + readout time + any user added delays.
 
A standard operational workflow is to:
Queue waveforms on galvos 1…N
Queue waveforms on remote focusing device.
Queue dc waveform for analog output of laser. Can be a vector of voltages, or a single voltage value.
Queue digital output of laser.
Queue external trigger of camera. Camera is edge triggered.
Software triggers all queued waveforms, causing them to roll off the device synchronously.
 
In a per-stack acquisition mode, where an entire channel is acquired for CH0, then CH1…, we would want to repeat this output for every image slice. In a per-z acquisition mode, where image slices are acquired in a channel interleaved fashion (CH0, CH1, step in z…), I would want to adjust the waveforms according to the user-specified settings (amplitude, offset, etc.) on a per-slice basis.
 
I’m not sure what the rate-limiting steps of such a process would be. For NI cards, writing the waveforms to the card are our slowest step. Thus, the per-z acquisition mode is frustratingly slow since we write data to tasks for every image. I have ideas on how to fix this, but I will see wha you come up with. 

Your job is to figure out how we can consolidate it, while ensuring that setting it up remains easy and obvious. One apparent solution is to use ASI's Tiger Controller, but other options are welcome too. Should you choose to proceed with this manufacturer/system, some information is below. 

*ASI Cards*
TGDAC4 has 4 analog outputs.  It does support some form or arbitrary waveforms with 250 points (and maybe extendable slightly more).  It also supports repeating patterns like sawtooth/sine.  It is 16 bit and has a 4kHz update rate.

TGPLC is 8 digital I/O ports.  It does not currently support arbitrary waveforms but there is the possibility that it could be implemented.  Due to hardware it would probably be able to get something like 1k or maybe up to 4k arbitrary points but not nearly what the NIDAQ cards achieve.  4kHz max update rate. For standard operation, this device pretty much supports ON/OFF gating, which can be used to execute pretty sophisticated triggering routines. Some examples of this can be found as a plugin to MicroManager (see the latest build). The PLC can also be setup using pymmcore_plus based off scripts at the bottom of the docs page.

AdvancedImagingUTSW, 2024-09-25:
Here is an example of a per channel z-stack acquisition. 

[image]

jondaniels, 2024-09-30:
I doubt ASI will be faster than NI at loading waveforms unless NI's design is really crappy, but maybe different acquisitions could be preloaded and played through later so that you batch the loading process.  The Tiger's baud rate sets at upper limit of ~11k bytes per second but I doubt that's achievable in practice because that the controller needs to acknowledge receipt of messages and there is other overhead.  .  I think a few thousand bytes per second transfer is realistic.  Let's optimistically guess that Tiger can accommodate transferring 4000 bytes per second and run some calculations based on that: if you were loading 250 points of analog in 4 channels that would take 0.5 second.  Or 1000 points of 8-bit digital output that would be 0.25 second.

ASI's analog outputs have an adjustable output filter that will help smoothing.  (There are 2 frequencies that can be set, each applies to one pair of outputs which I don't think is very clear in our documentation.)

@AdvancedImagingUTSW, How many points do you think it would take to re-create the most complicated analog waveforms required?  Related, what is that waveform's period?  (hopefully 4kHz output rate is enough).

AdvancedImagingUTSW, 2024-10-01:
Hello @jondaniels,

Thanks for the response. I think we could greatly accelerate writing waveforms to the NI cards if we were to decrease the size of the waveform. I'll benchmark it to see how fast we can write to NI tasks for different waveform sizes... I would be that a 250 point waveform is going to be pretty fast...

The very fastest waveform that we have to drive is the shadow reduction galvo. Ideally, this would be a resonant galvo, and we could just send it a constant voltage offset, but these galvos are harder to come by now... For our highest resolution ASLM, with linear galvos, this would ideally be close to ~200 Hz but with a small amplitude sine wave.

jondaniels, 2024-10-02:
For the resonant galvo you may not need to write a waveform at all.  The analog card supports outputing a sine wave by specifying the period, amplitude, and offset (also square, triangle, and sawtooth patterns with the same controls)  There is a way to trigger this output with an input TTL signal if free-running doesn't work.  This should work fine at 200Hz (with 4kHz update that gives 20 points per cycle).  See https://asiimaging.com/docs/singleaxis