TheDeanLab/navigate #1004: Evaluate Zarr V3

AdvancedImagingUTSW, 2024-10-01:
For Windows machines, appears to be a significant gain in write speed due to sharding implementation in Zarr V3. 

https://github.com/zarr-developers/zarr-python/pull/1111
https://zarr-specs.readthedocs.io/en/latest/v3/core/v3.0.html
https://forum.image.sc/t/ngff-ome-zarr-how-fast-can-you-write-it/74303/14

I think it requires that we set an environment variable to take the updated version. Need to check more recent versions of Zarr.

os.environ["ZARR_V3_EXPERIMENTAL_API"]="1"
See ...zarr/_storage/v3_storage_transformers.py

@zacsimile - Might be of interest to you as well.

AdvancedImagingUTSW, 2025-01-13:
Official announcement: https://forum.image.sc/t/zarr-python-3-is-here/107024

Also, Mark Kittisopikul provided a nice overview here: 

https://x.com/markkitti/status/1878146106955079981

```
This may seem confusing because having independently accessible chunks has been stated as the main advantage of Zarr over HDF5 or TIFF files. With poorly formatted HDF5 or TIFF files one often has to search through them to find the chunk locations. 

A key feature of Zarr v3 shards is that the chunk locations (chunk index) is aggregated into a single block either at the beginning or the end of the shard file. https://zarr-specs.readthedocs.io/en/latest/v3/codecs/sharding-indexed/v1.0.html

For builders of microscope systems, this is great news. You can take existing software that writes out z-stacks plane by plane, and then create a compatible Zarr shard by appending a chunk index to the file. In this case your the planes acquired from the camera are your chunks!

Preferably you would want to create 3D or perhaps even 4D (XYZC or XYZT) chunks at acquisition to optimize for future 3D or nD image processing or viewing. This can be done easily as by permuting dimensions of an array.

If you have a 1024 x 512 x 105 pixel z-stack, without permutation this can be reshaped into a 6D array as follows: (64 x 16) x (64 x 8) x (35 x 3).

Pemuting the dimensions rearranges the array to become (64 x 64 x 35) x (16 x 8 x 3) which can be saved as a shard directly.
```

mkitti, 2025-07-04:
Philipp Keller's lab uses Raid 0 NVME SSD arrays on acquisition computers. This allows us to perform acquisition at 5 GiB/second. The ability to do so primarily depends on the storage driver configuration. We have had a relatively good experience with software RAID.

Achieving this requires prudent use of system I/O calls. On Windows, most file I/O will eventually involve [`WriteFile`](https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-writefile) or [`WriteFileEx`](https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-writefileex). Generally, the idea is to minimize the number of calls to `WriteFile` and then get out of the way so the operating can do it's job.

This has two implications:
1. Write larger files. 2 GB is a practical size as this number of bytes can be described by a signed 32- bit integer.
2. Use asynchronous I/O. On Windows this is called [overlapped I/O](https://learn.microsoft.com/en-us/windows/win32/sync/synchronization-and-overlapped-input-and-output). Essentially this means we hand a pointer to a large buffer to Windows to write and then we do not wait for Windows to indicate it is finished.

In practice, this means you will want to create a circular buffer to receive camera frames. If you want to chunk by permuting the array as described in my X comment, you can either do that in place in the circular buffer or with another circular buffer. This buffer should be some multiple of the file size (i.e. 4 GB or 6 GB)

From the circular buffer, you offer a pointer to `WriteFile` with overlapped I/O. To synchronize before exiting, you can use an [I/O Completion Port](https://learn.microsoft.com/en-us/windows/win32/fileio/i-o-completion-ports).

Zarr v3 allows for efficient use of this system due to the [sharding codec](https://zarr.dev/zeps/accepted/ZEP0002.html). This allows us to write large files with individual chunks in it. In principle, you could have done this with tiled (Geo)TIFF or HDF5 as well, but the underlying file formats as slightly more complicated. The advantage that Zarr v3 shards here is that you can add the chunk index at the end of the file. Effectively, you could focus entirely on writing pixel data during acquisition and then add a few kilobytes to each file and a `zarr.json` to format the stacks as a Zarr v3 array. Since the format is relatively simple you do not need to rely on a 3rd party library such to write data. Instead, you can used the asynchronous overlapped I/O as described above.

mkitti, 2025-07-04:
If you want to hedge your bets, it is possible to create a file that is simultaneously a TIFF file, a HDF5 file, and a Zarr v3 shard.

https://github.com/mkitti/simple_image_formats/

To integrate this with the above, I create a 2 GB template of the combined file format. I either then write that filled in template or write data into a blank template at a known offset.

AdvancedImagingUTSW, 2025-07-07:
That is incredibly helpful @mkitti. Thank you!