TheDeanLab/navigate #1009: ASI stage connection error AND laser physical channel does not exist

EvoleneP, 2024-10-03:
**Describe the bug**
Lasers are not turning on with navigate. I also have an ASI connection error.

**Expected behavior**
On Wednesday 10-02-2024 we were able to control the lasers with navigate, only the power was controlled with Coherent software. I turned off all hardware but not the computer. When I turned everything back on this morning this happened. About to reboot hardware AND computer now to see if I can get it to behave again...

**Traceback**
(base) C:\Users\Nikon>conda activate navigate

(navigate) C:\Users\Nikon>navigate
2024-10-03 11:19:24,128 - model - ERROR - asi: ASI stage connection failed.
2024-10-03 11:19:24,628 - model - ERROR - asi: ASI stage connection failed.
LaserBase error:, Physical channel specified does not exist on this device.
Refer to the documentation for channels available on this device.
Device: PCI6321
Physical Channel Name: ao11

Task Name: _unnamedTask<1>

Status Code: -200170, DAQmxErrors.PHYSICAL_CHAN_DOES_NOT_EXIST, -200170
LaserBase error:, Physical channel specified does not exist on this device.
Refer to the documentation for channels available on this device.
Device: PCI6321
Physical Channel Name: ao11

Task Name: _unnamedTask<5>

Status Code: -200170, DAQmxErrors.PHYSICAL_CHAN_DOES_NOT_EXIST, -200170
LaserBase error:, Physical channel specified does not exist on this device.
Refer to the documentation for channels available on this device.
Device: PCI6321
Physical Channel Name: ao2

Task Name: _unnamedTask<7>

Status Code: -200170, DAQmxErrors.PHYSICAL_CHAN_DOES_NOT_EXIST, -200170

**Logs & Configuration Files**
2024-10-03 11:19:24,116 - model - INFO - decorators: DAQBase, (<DictProxy object, typeid 'dict' at 0x125ffba23a0>,), {}
2024-10-03 11:19:24,116 - model - INFO - decorators: NIDAQ, (<DictProxy object, typeid 'dict' at 0x125ffba23a0>,), {}
2024-10-03 11:19:24,128 - model - ERROR - asi: ASI stage connection failed.
2024-10-03 11:19:24,628 - model - ERROR - asi: ASI stage connection failed.
2024-10-03 11:19:25,197 - model - INFO - base: HamamatsuOrca, Offset or variance map not found in C:\Users\Nikon\AppData\Local\.navigate\camera_maps
2024-10-03 11:19:25,197 - model - INFO - decorators: CameraBase, ('OPM', <navigate.model.devices.APIs.hamamatsu.HamamatsuAPI.DCAM object at 0x00000125BD367EE0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,464 - model - INFO - decorators: HamamatsuBase, ('OPM', <navigate.model.devices.APIs.hamamatsu.HamamatsuAPI.DCAM object at 0x00000125BD367EE0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,464 - model - INFO - decorators: HamamatsuOrca, ('OPM', <navigate.model.devices.APIs.hamamatsu.HamamatsuAPI.DCAM object at 0x00000125BD367EE0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,485 - model - INFO - decorators: RemoteFocusBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,485 - model - INFO - decorators: SyntheticRemoteFocus, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,507 - model - INFO - decorators: GalvoBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:25,507 - model - INFO - decorators: SyntheticGalvo, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:25,519 - model - INFO - decorators: FilterWheelBase, (<navigate.model.devices.APIs.asi.asi_tiger_controller.TigerController object at 0x00000125BD367DC0>, <DictProxy object, typeid 'dict' at 0x125bdf7ad60>), {}
2024-10-03 11:19:25,529 - model - INFO - decorators: ASIFilterWheel, (<navigate.model.devices.APIs.asi.asi_tiger_controller.TigerController object at 0x00000125BD367DC0>, <DictProxy object, typeid 'dict' at 0x125bdf7ad60>), {}
2024-10-03 11:19:25,546 - model - INFO - decorators: ZoomBase, ('OPM', <navigate.model.device_startup_functions.DummyDeviceConnection object at 0x00000125BD367340>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,547 - model - INFO - decorators: SyntheticZoom, ('OPM', <navigate.model.device_startup_functions.DummyDeviceConnection object at 0x00000125BD367340>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,555 - model - INFO - decorators: ShutterBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,556 - model - INFO - decorators: SyntheticShutter, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>), {}
2024-10-03 11:19:25,571 - model - INFO - decorators: LaserBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:25,738 - model - INFO - decorators: LaserNI, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:25,748 - model - INFO - decorators: LaserBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 1), {}
2024-10-03 11:19:25,761 - model - INFO - decorators: LaserNI, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 1), {}
2024-10-03 11:19:25,770 - model - INFO - decorators: LaserBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 2), {}
2024-10-03 11:19:25,781 - model - INFO - decorators: LaserNI, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 2), {}
2024-10-03 11:19:25,790 - model - INFO - decorators: LaserBase, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 3), {}
2024-10-03 11:19:25,801 - model - INFO - decorators: LaserNI, ('OPM', None, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 3), {}
2024-10-03 11:19:25,825 - model - INFO - decorators: StageBase, ('OPM', <navigate.model.devices.APIs.asi.asi_tiger_controller.TigerController object at 0x00000125BD367DC0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:26,844 - model - INFO - decorators: ASIStage, ('OPM', <navigate.model.devices.APIs.asi.asi_tiger_controller.TigerController object at 0x00000125BD367DC0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 0), {}
2024-10-03 11:19:26,867 - model - INFO - decorators: StageBase, ('OPM', <navigate.model.devices.daq.ni.NIDAQ object at 0x00000125BD367D90>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 1), {}
2024-10-03 11:19:26,910 - model - INFO - decorators: GalvoNIStage, ('OPM', <navigate.model.devices.daq.ni.NIDAQ object at 0x00000125BD367D90>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 1), {}
2024-10-03 11:19:26,928 - model - INFO - decorators: StageBase, ('OPM', <navigate.model.device_startup_functions.DummyDeviceConnection object at 0x00000125BDF64CA0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 2), {}
2024-10-03 11:19:26,928 - model - INFO - decorators: SyntheticStage, ('OPM', <navigate.model.device_startup_functions.DummyDeviceConnection object at 0x00000125BDF64CA0>, <DictProxy object, typeid 'dict' at 0x125ffba23a0>, 2), {}
2024-10-03 11:19:27,369 - model - INFO - base: HamamatsuOrca, Offset or variance map not found in C:\Users\Nikon\AppData\Local\.navigate\camera_maps
2024-10-03 11:19:27,518 - model - INFO - model: Change zoom of OPM to 47.6x
2024-10-03 11:19:27,552 - model - INFO - microscope: Microscope state: {'microscope_name': 'OPM', 'image_mode': 'live', 'zoom': '47.6x', 'stack_cycling_mode': 'per_stack', 'start_position': -100.0, 'end_position': 100.0, 'step_size': 10.0, 'number_z_steps': 20, 'timepoints': 3, 'stack_pause': 0.0, 'is_save': False, 'stack_acq_time': 1.0, 'timepoint_interval': 0, 'experiment_duration': 3.03, 'is_multiposition': False, 'multiposition_count': 1, 'selected_channels': 1, 'channels': <DictProxy object, typeid 'dict' at 0x125bdfa4c40>, 'stack_z_origin': 110, 'stack_focus_origin': 1907, 'start_focus': 0.0, 'end_focus': 0.0, 'abs_z_start': 10.0, 'abs_z_end': 210.0, 'scanrange': 500.0, 'n_plane': 1.0, 'offset_start': 0.0, 'offset_end': 10.0, 'conpro_cycling_mode': 'per_plane', 'waveform_template': 'Default', 'autofocus_device': 'stage', 'autofocus_device_ref': 'f'}
2024-10-03 11:19:27,555 - model - INFO - microscope: Waveform constants: {'remote_focus_constants': <DictProxy object, typeid 'dict' at 0x125bdfa4c10>, 'galvo_constants': <DictProxy object, typeid 'dict' at 0x125bdfa4b80>, 'other_constants': <DictProxy object, typeid 'dict' at 0x125bdfa4ca0>}
2024-10-03 11:19:27,660 - model - INFO - microscope: Stage limits enabled: False

AdvancedImagingUTSW, 2024-10-03:
Hi @EvoleneP, I am convinced that the ASI stage issue is unrelated to our software and is a problem with your Tiger Controller.

The other error is because your configuration.yaml file specifies analog output ports that do not exist. According to the pinout of the PCIe-6321, this device only has AO0 and AO1. I believe this is configured for the analog modulation of the lasers. Is that right? Are these errors crashing the program, or are they just being logged?

I would change the configuration.yaml file to have your two most important lasers as AO0 and AO1 for their analog outputs.

closed 2024-10-03 by AdvancedImagingUTSW

EvoleneP, 2024-10-03:
Hi Kevin

Yes, the ASI issue is unrelated I agree, it just so happened after not happening in a while...

As for the lasers, I don't know what to tell you. I didn't change anything since yesterday when we had it working when we changed the configuration file other than turning off the hardware for the night. I changed the configuration file back to digital only and been using the Coherent software to turn on the lasers and control their power as before. I tried to reproduce the steps to get the lasers to be controlled by navigate like you did yesterday but nothing worked. I can keep working but ultimately it would be nice to get the software to work with everything, especially when we finally get users. Maybe after the workshop when you have more time we can try again?

Best

Evolene Premillieu, PhD
Light Sheet Microscopy specialist
BioFrontiers Institute
University of Colorado Boulder
https://advancedimaging.colorado.edu/

[External email - use caution]

AdvancedImagingUTSW, 2024-10-04:
Hello @EvoleneP,

While the configuration file didn't change, how you ran the software probably did. I assume that when you and I tested it together, we didn't select the channel that was configured on AO2 or AO11. We were probably lucky and confirmed that the lasers were working with AO0 and AO1. 

We will fix the issue where if the analog modulation is set to synthetic, the digital modulation does not work. However, in the long-run, you will not be able to module the intensity of the laser with the analog output using that card; it doesn't have enough analog outputs. You have 4 lasers, and it has 2 analog outputs.

Another solution is to enable software control of the laser intensity (e.g., via USB). We have the code to do this, but it hasn't been a priority to integrate it.

So, for the time being, you will have to control the intensity of the lasers from within the Obis software, and use the digital outputs to trigger then on and off (pending #1008).