TheDeanLab/navigate #1038: Installing navigate and navigate MMcore

jcknowles, 2024-11-25:
I am not a programmer so have been tryin to step through the instructions. 

Ive installed miniconda
Got this running. 
Installed Navigate.
Using C:\users\lighsheet folder
Downloaded (is this the same as cloning?) MMCore zip file and unzipped it to C:\users\lighsheet\navigate-mmcore-plugin-develop folder
Configured the windows environment

If I run navigate as navigate -sh it starts

If I run just navigate i get the following errors

vvvvvvvvvvvvvvvvvvvvvvvvvvv Child Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvv
 Traceback (most recent call last):
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 501, in _child_loop
    obj = initializer(*initargs, **initkwargs)
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\model.py", line 115, in __init__
    devices_dict = load_devices(
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\device_startup_functions.py", line 1393, in load_devices
    camera = load_camera_connection(configuration, id, is_synthetic)
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\device_startup_functions.py", line 172, in load_camera_connection
    HamamatsuController = importlib.import_module(
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\devices\APIs\hamamatsu\HamamatsuAPI.py", line 38, in <module>
    __dll = windll.LoadLibrary("dcamapi.dll")
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\ctypes\__init__.py", line 460, in LoadLibrary
    return self._dlltype(name)
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\ctypes\__init__.py", line 382, in __init__
    self._handle = _dlopen(self._name, mode)
FileNotFoundError: Could not find module 'dcamapi.dll' (or one of its dependencies). Try using the full path with constructor syntax.
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^ Child Process Traceback ^^^^^^^^^^^^^^^^^^^^^^^^^^^
 vvvvvvvvvvvvvvvvvvvvvvvvvvv Main Process Traceback vvvvvvvvvvvvvvvvvvvvvvvvvvvv
Traceback (most recent call last):
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\Scripts\navigate.exe\__main__.py", line 7, in <module>
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\main.py", line 113, in main
    Controller(
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\controller\controller.py", line 195, in __init__
    self.model = ObjectInSubprocess(
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 383, in __init__
    assert _get_response(self) == "Successfully initialized"
  File "C:\Users\Lightsheet\miniconda3\envs\navigate\lib\site-packages\navigate\model\concurrency\concurrency_tools.py", line 438, in _get_response
    raise resp
FileNotFoundError: Could not find module 'dcamapi.dll' (or one of its dependencies). Try using the full path with constructor syntax.

jcknowles, 2024-11-25:
> ⚠ Caution: External sender
>
> Thanks for not hesitating to reach out. Could you also share with us your configuration file from your .navigate folder? I assume navigate opens if you give the command navigate -sh?
>
> At first glance, it appears that you need to install the Hamamatsu drivers. You should make sure that you are able to open your camera using their software.
>
> https://www.hamamatsu.com/us/en/product/cameras/software/driver-software.html
>
> Let me know how it goes and we can connect via Zoom to get things up and working.
>
> Best,
> Kevin

Many thanks for the rapid reply!

Maybe before getting further I should explain what we are trying to do and if navigate could do this

We built an openspim lightsheet with an omicron laser, Picard stage and PCO camera

We have found the openspim software crashes a lot so were trying to see if there an alternative and found navigate.

Would it be able to control the above hardware or not?

Thanks

Jonathan

[signature_3768372275]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

AdvancedImagingUTSW, 2024-11-25:
Hello Johnathan,

I just quickly reviewed the parts list for the OpenSPIM, and we can get you most of the way there. It ultimately depends upon your timeline and willingness to purchase another piece of equipment. 

Navigate uses National Instruments data acquisition cards to perform all analog and digital triggering of cameras and lasers. It does not seem like the OpenSPIM requires this. Would you happen to have one of these cards lying around? They run around $3k USD (see attached quote from Newark). Additionally, we would have to add support for the PCO camera, but I don't suspect that would be too difficult. The PCO camera has a great API, and we know several labs who routinely control these cameras with their own Python software.

Otherwise, the Picard stage is supported my the MMCore plugin, and @annie-xd-wang would be able to help you with that. And most likely, we could wire up the lasers for analog/digital external control no problem. What exact model of Omicron do you have? We typically use the LightHUB systems...

All the best,
Kevin

[Newark[41].pdf](https://github.com/user-attachments/files/17903695/Newark.41.pdf)

jcknowles, 2024-11-25:
Hi Again

We do have some NI hardware lying around I will check.

We actually trigger our laser via an Arduino which works in Micromanager. Would this work instead?

I will check the boxes for NI cards

Jonathan

[signature_384482827]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

jcknowles, 2024-11-25:
Weve got a USB I/O unit. The nearest PCIe card we have is a 6320

https://www.newark.com/ni/781043-01/pcie-6320-multifun-i-o-device/dp/13AJ3954

Jonathan

[signature_1673973874]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

AdvancedImagingUTSW, 2024-11-25:
The PCIe card would work for triggering cameras, shutters, and digital modulation of lasers. It does not have analog outputs, so laser intensity would have to be set using the Omicron software. 

There are no galvos, right? It is a sample scanning system with a static light-sheet? 

[image]

jcknowles, 2024-11-25:
Hi

The system is just a Move sample/turn laser on/take image/turn laser off/move to next position.

As I mentioned the laser control is currently done via the Arduino using a TTL output. We only have one wavelength at the moment

Thanks

Jonathan

[signature_3276897447]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

image.png (view on web)<https://github.com/user-attachments/assets/9232d58f-e1d3-4c95-99e3-be30c2a510b8>

AdvancedImagingUTSW, 2024-11-25:
That sounds simple enough. What is the exact model of the PCO camera that you have? That is the last thing holding us up...

Best,
Kevin

jcknowles, 2024-11-25:
From memory it’s a pco edge 4.2 usb3. Would that be correct? Am on the way home so can’t check!

Professor J.C. Knowles BSc(hons), PhD, FAIMBE, FIMMM, CEng, FRSC, CSci
Professor of Biomaterials Science
Head of Division of Biomaterials and Tissue Engineering,
UCL Eastman Dental Institute,
University College London,
Royal Free Hospital, Rowland Hill Street London NW3 2PF. UK.
Telephone  +44 020 8016 7699
Mobile 07785 313615
Editor, Journal of Tissue Engineering
Editor, Journal of Biomaterials Applications
Email ***@***.******@***.***>
Eastman website; http://www.ucl.ac.uk<http://www.ucl.ac.uk/>/eastman
Personal website; http://www.homepages.ucl.ac.uk/~sfhvjck/

AdvancedImagingUTSW, 2024-11-25:
Sounds good. Thanks for the information. Before you do anything, the first thing we need to do is evaluate how difficult it will be for us to add software for controlling that PCO camera. Please let us know what your timeline is. It won't be overnight on our side to add new code for controlling the camera.

jcknowles, 2024-11-25:
Not in any rush with this thanks
Professor J.C. Knowles BSc(hons), PhD, FAIMBE, FIMMM, CEng, FRSC, CSci
Professor of Biomaterials Science
Head of Division of Biomaterials and Tissue Engineering,
UCL Eastman Dental Institute,
University College London,
Royal Free Hospital, Rowland Hill Street London NW3 2PF. UK.
Telephone  +44 020 8016 7699
Mobile 07785 313615
Editor, Journal of Tissue Engineering
Editor, Journal of Biomaterials Applications
Email ***@***.******@***.***>
Eastman website; http://www.ucl.ac.uk<http://www.ucl.ac.uk/>/eastman
Personal website; http://www.homepages.ucl.ac.uk/~sfhvjck/

jcknowles, 2024-11-26:
Hi

I will ask around at UCL and see if we could borrow one of the NI cards. Would this make things easier?

For the laser in micromanager, we could control the power via the USB connection and just switched it on and off via the Arduino

Thanks

Jonathan

[signature_2721750378]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

AdvancedImagingUTSW, 2024-11-26:
Hello Jonathan,

The NI card you have is perfectly fine. We can control the power of the lasers externally (either via MicroManager or with Omicron's software), and use the NI card to trigger the laser and the camera. Of course, you would need a cable and breakout box to connect to your PCIe-based NI card. We would not use the Ardruino.  Annie is looking into the PCO camera and we hope to get back you fairly quickly. 

Best,
Kevin

jcknowles, 2024-11-26:
Thanks for all the help with this.

I think we have a breakout box for the board already. Otherwise I will get one. Also I will install the drivers and card. Anything specific software wise needed for the NI card as I know NI supply a lot of packages with their cards.

Professor J.C. Knowles BSc(hons), PhD, FAIMBE, FIMMM, CEng, FRSC, CSci
Professor of Biomaterials Science
Head of Division of Biomaterials and Tissue Engineering,
UCL Eastman Dental Institute,
University College London,
Royal Free Hospital, Rowland Hill Street London NW3 2PF. UK.
Telephone  +44 020 8016 7699
Mobile 07785 313615
Editor, Journal of Tissue Engineering
Editor, Journal of Biomaterials Applications
Email ***@***.******@***.***>
Eastman website; http://www.ucl.ac.uk<http://www.ucl.ac.uk/>/eastman
Personal website; http://www.homepages.ucl.ac.uk/~sfhvjck/

jcknowles, 2024-11-27:
Hi

Ive found a breakout box and cable. Its an NI SCC-68

Jonathan

[signature_2240095356]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

AdvancedImagingUTSW, 2024-11-27:
That looks like it will work. The pinout is here:
https://www.ni.com/docs/en-US/bundle/scc-68-seri/resource/374748b.pdf

jcknowles, 2024-11-28:
Thanks.

Ive installed the card and DAQmx and the card has installed OK and is detected by the NI software

The box is a type M series and looking on p39 I am assuming we need an analog out so pin 22 and 54.

Thanks again
Jonathan

[signature_4106210862]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>

https://www.ni.com/docs/en-US/bundle/scc-68-seri/resource/374748b.pdf

AdvancedImagingUTSW, 2025-01-28:
@jcknowles - I have finally gotten in touch with someone from PCO about getting a loaner camera so that we can implement the code. Before I finalize the request, I wanted to make sure that you are still interested.

jcknowles, 2025-01-29:
Hi

Weve been back to Openspim and we seem to have realised why the system keeps crashing. It needs time to “settle” between commands/functions.

I guess the question is about the differences between navigate and openspim? If they offer similar functionalities then yes we would be interested in Navigate as an alternative. However is openspim offers more functionality then we might just live with it and its quirls.

Does this make sense?

Thanks

Jonathan

[signature_837797437]<http://www.ucl.ac.uk/eastman>

Professor Jonathan Knowles, BSc, PhD, FIMMM, FRSC, CEng, FAIMBE
Professor of Biomaterials Science
Division of Biomaterials and Tissue Engineering
UCL Eastman Dental Institute
Royal Free Hospital, Rowland Hill Street,
London, NW3 2PF, UK
Tel: +44 (0)20 8016 7699
My Profile<https://profiles.ucl.ac.uk/5968-jonathan-knowles/about>
Editor in Chief: Journal of Biomaterials Applications<https://journals.sagepub.com/home/jba>
Editor in Chief: Journal of Tissue Engineering<https://journals.sagepub.com/home/tej>
To stay up to date with the latest Institute news:
visit the website<http://www.ucl.ac.uk/eastman> | sign up for emails<http://www.ucl.ac.uk/eastman/about/sign-up-for-emails> | follow us on Twitter<http://www.twitter.com/ucleastman>