TheDeanLab/navigate #1080: More Abstract Stage Definition

AdvancedImagingUTSW, 2025-03-24:
For @conorhughmcfadden ...

Bare minimum requirements:
Ability to specify additional stage axes in configuration file, get the appropriate widget in the GUI, and add additional columns in the multi-position table. Additional widgets can at least initially show up in a 4th column.

[image]

Future additions could include ability to tile over this - thus being in the tiling wizard, and the ability to specify additional stage axes in the configurator. For now, let's not worry about these things.

annie-xd-wang, 2025-03-26:
The bare minimum requirements have been implemented in the [PR1083](https://github.com/TheDeanLab/navigate/pull/1083).

AdvancedImagingUTSW, 2025-03-27:
I quickly tested this by adding 5 additional stages, and of course, it works great. @conorhughmcfadden Why don't you try to configure an additional stage?

[image]

[image]

closed 2025-04-02 by AdvancedImagingUTSW