TheDeanLab/navigate #1084: Error on starting navigate, does not open

EvoleneP, 2025-03-26:
OUt fo the blue I am having this error when starting navigate. The Logo window appears then this error message shows up. I restarted all the hardware and updated navigate and it still happens.
Will try a full PC + hardware reboot and see if anything changes...

(base) C:\Users\Nikon>conda activate navigate

(navigate) C:\Users\Nikon>navigate
Traceback (most recent call last):
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Nikon\miniconda3\envs\navigate\Scripts\navigate.exe\__main__.py", line 7, in <module>
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\main.py", line 114, in main
    Controller(
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 338, in __init__
    self.populate_experiment_setting(in_initialize=True)
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\controller.py", line 484, in populate_experiment_setting
    self.camera_setting_controller.populate_experiment_values()
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\camera_settings.py", line 260, in populate_experiment_values
    self.update_fov()
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\camera_settings.py", line 511, in update_fov
    self.calculate_physical_dimensions()
  File "C:\Users\Nikon\Documents\GitHub\navigate\src\navigate\controller\sub_controllers\camera_settings.py", line 588, in calculate_physical_dimensions
    pixel_size = self.parent_controller.configuration["configuration"][
  File "<string>", line 2, in __getitem__
  File "C:\Users\Nikon\miniconda3\envs\navigate\lib\multiprocessing\managers.py", line 825, in _callmethod
    raise convert_to_error(kind, result)
KeyError: '51x'

annie-xd-wang, 2025-03-26:
It looks like the pixel size information for 51x zoom is missing. Could you add this to the configuration.yaml? It should look like the following lines:

    zoom:
      hardware:
        type: DynamixelZoom # your device type
        servo_id: 1
        port: COM9
        baudrate: 100000
      position:
        1x: 627
        51x: 1711
      pixel_size:
        1x: 6.38
        51x: 1.044

EvoleneP, 2025-03-26:
Hi Annie

Thank you, I will try this!

Best

Evolene Premillieu, PhD
Light Sheet Microscopy specialist
BioFrontiers Institute
University of Colorado Boulder
https://advancedimaging.colorado.edu/

[External email - use caution]

EvoleneP, 2025-04-03:
I think I solved my issue, software opens now.

closed 2025-04-03 by EvoleneP