TheDeanLab/navigate #1097: Projection Mode Imaging: Can't ZStack Afterwards/Task Issues in NIStage

conorhughmcfadden, 2025-06-02:
**Describe the bug**

I'm trying to incorporate a feature to do Projection Imaging; we want to z-galvo to sweep over each frame at some amplitude. See the implementation as a `common_feature` here on my branch;

[navigate/tree/projection-mode](https://github.com/TheDeanLab/navigate/tree/projection-mode)

It works as a feature using the `enable` variable as a toggle; `True` turns it on and `False` should turn it off. I do it this way to make it flexible, i.e) I put a ProjectionMode(enable=True) block before TonyWilson to do Projection-AO, and then I ProjectionMode(enable=False) to set the zGalvo back to it's native state.

The problem is that it doesn't really go back. I tried `switch_mode("normal")` and setting waveform template to Default, but it seems there is some issue with the task still being reserved. 

In `stages/ni.py`, when you call `update_waveform` for the galvo, `switch_mode("waveform")` is called. All this does, AFAIK is kill the task. I'm sure there is a reason we're doing this, but I think it may be causing some of my issues.

**To Reproduce**

I test it as follows. I run the following feature list:

```console
[{"name": ProjectionMode,"args": ("z",0,False,),},{"name": PrepareNextChannel,},{"name": ZStackAcquisition,"args": (False,False,"z-stack",False,),},{"name": WaitToContinue,},]
```

It just does a ZStack. Works great with no issues.

Now I run this second feature list which sets up Projection imaging and does TonyWilson AO:

```console
[{"name": ProjectionMode,"args": ("z",0,True,),},{"name": PrepareNextChannel,},{"name": TonyWilson,},]
```

Again, works great! I see projection mode beads and AO. Fantastic!

Now I run the first one again:

```console
[{"name": ProjectionMode,"args": ("z",0,False,),},{"name": PrepareNextChannel,},{"name": ZStackAcquisition,"args": (False,False,"z-stack",False,),},{"name": WaitToContinue,},]
```

The stack runs, but the galvo doesn't move. I just collect 10 or so frames while the z galvo is parked at zero. You can tell the amplitude of the z-galvo is also wrong, it is still sweeping like in projection mode, where it should be going stepwise with each slice. However, it does finish running without hanging.

**Expected behavior**

I would like ProjectionMode(enable=False) to set the z-galvo back to it's original state, so that once I've done Projection AO I can acquire a stack normally, without having to mess around switching between config files or different microscopes.

**Logs & Configuration Files**

I'm using the following config:

```yaml
# Only one microscope can be active in the GUI at a time, but all microscopes will be accessible
microscopes:
  OPM-VAST:
    daq:
      hardware:
        name: daq
        type: NI

      # NI PCIe-1073 Chassis with PXI-6259 and PXI-6733 DAQ Boards.
      # Sampling rate in Hz
      sample_rate: 100000
      sweep_time: 0.2

      # triggers
      master_trigger_out_line: /PCIe-6738-A/port0/line1
      camera_trigger_out_line: /PCIe-6738-A/ctr0 #PFI7 Camera trigger
      trigger_source: /PCIe-6738-A/PFI0

    mirror:
      hardware:
        name: mirror
        type: ImagineOpticsMirror
        flat_path: D:\WaveKitX64\MirrorFiles\FlatReference.wcs
      n_modes: 32

    zoom:
      hardware:
        name: zoom
        type: synthetic
        servo_id: 1
      position:
        1x: 0
      pixel_size:
        1x: 0.15
    
    shutter:
      hardware:
        name: shutter
        type: synthetic
        channel: none/line0
      shutter_min_do: 0
      shutter_max_do: 5
    
    camera:
      hardware:
        type: Photometrics
        serial_number: 111
        camera_connection: PMPCIECam00
      pixel_size_in_microns: 0.1483 # 25X, NA1.10, OPMv3 measured 09/29/2024 (C:\Users\vastopmv3\Desktop\ASLMData\Conor\PhotometricsBuild\Grid\PixelSize\2024-09-29\Cell_001)
      readout_port: 2
      gain: 1
      speed_table_index: 0
      defect_correct_mode: 2.0
      delay: 1.0  #ms
      settle_down: 0.1 #ms
      flip_x: False
      flip_y: False
      unitforlinedelay: 1

    remote_focus_device:
      hardware:
        name: daq
        type: synthetic
        channel: none
        min: 0
        max: 5
      delay_percent: 0
      ramp_rising_percent: 50
      ramp_falling_percent: 2.5
      amplitude: -0.5
      offset: 0
    
    galvo:
      -
        # shear galvo measured: 351.04 um/V
        name: sheargalvo
        hardware:
          name: daq
          type: NI
          channel: PCIe-6738-A/ao12
          min: -5
          max: 5
        waveform: sawtooth
        frequency: 0.5
        amplitude: -1
        offset: 0
        duty_cycle: 50
        phase: 1.57079

    stage:
      hardware:
        -
          name: vast_stage
          # type: vast
          type: synthetic
          serial_number: 455
          axes: [x,y,theta]
          axes_mapping: [x,y,theta]
        -
          name: asi_z_stage
          type: MS2000
          # type: synthetic
          serial_number: 001
          axes: [m]
          axes_mapping: [Z]
          port: COM3
          baudrate: 9600
          timeout: 0.25          
        - 
          name: z_galvo
          type: GalvoNIStage
          # type: synthetic
          serial_number: 125
          axes: [z]
          axes_mapping: [PCIe-6738-A/ao0]
          min: -3.5
          max: 3.5
          volts_per_micron: 0.007*x
        - 
          name: snouty_piezo
          type: GalvoNIStage
          serial_number: 124
          axes: [f]
          axes_mapping: [PCIe-6738-A/ao21]
          min: 0
          max: 10
          volts_per_micron: (10/15.4)*x + 5.0      

      y_unload_position: 10000
      y_load_position: 90000

      startfocus: 0
      x_max: 5000
      x_min: -5000
      y_max: 1000
      y_min: -1000
      z_max: 1000  # z-stack galvo
      z_min: -1000
      m_max: 10000  # global z-stage (ASI MS-2000)
      m_min: -210000      
      f_max: 15 # snouty piezo 
      f_min: -15
      theta_max: 180
      theta_min: -180

      x_rot_position: 2000
      y_rot_position: 2000
      z_rot_position: 2000

      x_step: 50
      y_step: 50
      z_step: 50
      m_step: 50
      theta_step: 30
      f_step: 1

      position:
        x_pos: 5000
        y_pos: 5000
        z_pos: 25000
        f_pos: 0
        m_pos: 0
        theta_pos: 0
      velocity: 500

      x_offset: 0
      y_offset: 0
      z_offset: 0
      f_offset: 0
      m_offset: 0
      theta_offset: 0
    
    filter_wheel:
      hardware:
        name: filter_wheel
        type: LUDLFilterWheel
        port: COM1
        baudrate: 9600
        wheel_number: 1
      filter_wheel_delay: .050 # in seconds
      available_filters:
        530/43: 1
        525/50: 2
        560/94: 3
        600/52: 4
        561/LP: 5
        ND=4.0: 6

    lasers:
      # Omicron LightHub Ultra
      # 488 and 640 are LuxX+ Lasers
      # 561 is a Coherent OBIS Laser
      # Digital Laser Outputs
      - wavelength: 488
        onoff:
          hardware:
            name: daq
            type: NI
            channel: PCIe-6738-A/port1/line2
            min: 0
            max: 5
        power:
          hardware:
            name: daq
            type: NI
            channel: PCIe-6738-A/ao5
            min: 0
            max: 5
        type: Obis
        index: 0
        delay_percent: 10
        pulse_percent: 100
      - wavelength: 561
        onoff:
          hardware:
            name: daq
            type: NI
            channel: PCIe-6738-A/port1/line3
            min: 0
            max: 5
        power:
          hardware:
            name: daq
            type: NI
            channel: PCIe-6738-A/ao11
            min: 0
            max: 5
        type: synthtic # Obis
        index: 1
        delay_percent: 10
        pulse_percent: 100

gui:
  channels:
    count: 5
    laser_power:
      min: 0
      max: 100
      step: 10
    exposure_time:
      min: 1
      max: 1000
      step: 5
    interval_time:
      min: 0
      max: 1000
      step: 5
  stack_acquisition:
    step_size:
      min: 0.200
      max: 1000
      step: 0.1
    start_pos:
      min: -5000
      max: 5000
      step: 1
    end_pos:
      min: -5000
      max: 10000
      step: 1
  timepoint:
    timepoints:
      min: 1
      max: 1000
      step: 1
    stack_pause:
      min: 0
      max: 1000
      step: 1
  confocal_projection:
    scanrange:
      min: 0
      max: 600
      step: 1
    offset_start:
      min: -300
      max: 300
      step: 0.1
    offset_end:
      min: -300
      max: 300
      step: 0.1
    n_plane:
      min: 1
      max: 200
      step: 1
```

And here is the .log I get when running the above workflow:

[model_debug.log](https://github.com/user-attachments/files/20559726/model_debug.log)

AdvancedImagingUTSW, 2025-06-03:
@annie-xd-wang - if there is an easy fix or piece of advice you could provide to Conor on this, we would appreciate it.

AdvancedImagingUTSW, 2025-06-11:
@conorhughmcfadden - Have you been able to test this?