TheDeanLab/navigate #1106: Sidebar layout overflow in ‘Slices’ display mode

juhelh, 2025-06-26:
**Describe the bug**
When switching from "Live" to "Slices" in the Camera view, the right-hand sidebar (which contains LUT and display options) begins to stretch vertically, about one step per second. This quickly ruins the layout.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Navigate and aquire some images (I ran the synthetic hardware version with "navigate -sh", used continuous scan, pressed "Aquire" and then "Stop" shortly after).
2. In the “Image Display” dropdown, change from Live to Slices.
3. Observe the sidebar on the right-hand side (LUT, Autoscale, etc.).
4. The UI expands vertically over time.

**Expected behavior**
Sidebar layout should remain fixed in height regardless of the display mode.

**Traceback**
No relevant traceback appeared when the bug occurred.

**Logs & Configuration Files**
Attached: Log files 2025-06-26-1138.zip from a session run with synthetic hardware (Navigate 0.0.8). 
Attached: Configuration files config.zip. Reflects the settings at the time the issue occurred, though older sessions seem to be overwritten by default. 

[config.zip](https://github.com/user-attachments/files/20923734/config.zip)

[2025-06-26-1138.zip](https://github.com/user-attachments/files/20923748/2025-06-26-1138.zip)

AdvancedImagingUTSW, 2025-06-26:
@juhelh - Thank you for opening this issue. We will revisit the logic that enables us to have a flexible window size, which when incorrect, results in the error you observed.

AdvancedImagingUTSW, 2025-07-02:
Hello @juhelh - @annie-xd-wang adjusted the logic in PR #1108. If you are willing to test it out and make sure that it works on your system, we would greatly appreciate it. On our side, it looks great, but I want to rule out any possibilities.

closed 2025-07-02 by AdvancedImagingUTSW

juhelh, 2025-07-03:
Hi @AdvancedImagingUTSW! Just checked the PR, works well for me too now, thanks! 

Sidenote: Seems like you also did something with the scaling of the window (like making sure you can't shrink it below a point). Works fine. I guess an optional improvement would be making sure that when you increase the size all the widgets inside the window get scaled accordingly? Right now it only seems to add extra grey area to the right of camera view. Just a thought!